bdring/Grbl_ESP32_MPCNC_Controller #3: powered makerbot style endstop support

thawkins, 2019-09-22:
If you find yourself laying out a new revision, can you add support for powered end stops, these are very common on 3D printers and can be had for pennies, they include all the filters at the switch end and have an LED for confirming operation. 

see: http://marlinfw.org/assets/images/docs/hardware/endstops/makerbot_endstop.png

The only requirement, is to add an extra +5v pin on the end-stop pin-outs, so that each end-stop row has 

[Signal] - [Grnd] - [Vcc]  as per the RAMPS cards

Typical prices are very very low. 

here is a listing from my region, but these devices can be found everywhere

https://www.lazada.com.ph/products/6pcs-mechanical-endstop-limit-switch-press-switch-module-for-3d-printer-makerbot-prusa-mendel-reprap-cnc-arduino-mega-2560-1280-ramps-14-lkb01-i303860525-s555784736.html?spm=a2o4l.searchlist.list.29.6aa439896l4hTo&search=1

the price here for 6 pre-wired end-stops is about $3

bdring, 2019-09-22:
Have you tried using those with 3.3v? The resistor is pretty high, so they might not light up. 5V will kill an ESP32

thawkins, 2019-09-22:
I will have a play next week i have a couple of esp32, i can breadboard on 3.3v with these switches  if i can hook one up to a gpio and see if i can read them with a 3.3v Vcc.  I will also be able to see what the led briteness is like.

bdring, 2019-09-22:
I have used these switches. They work fine for close on contact if you use the red and white wires. They will light up at 3.3V. The resistor appears to be 810R, but very hard to read.

[They are common on amazon](https://www.amazon.com/gp/product/B07PCN6T6F)

[image]

thawkins, 2019-10-17:
I have confirmed that these style endstop boards work fine at 3.3v, in fact i cant really tell the difference between 5v and 3.3v operation, the leds seem equaly bright.