bdring/Grbl_ESP32_MPCNC_Controller #5: Non-MPCNC use

vertiginous, 2019-11-05:
Apologies if this isn't the right place to ask this question.

I want to use grbl_esp32 to control an openbuilds ox CNC router that I'm building.  It will need a single driver for x and z, and 2 drivers for Y.  Also, not sure it's relevant, but I plan to use TMC2130 drivers, as I believe I'll need the additional current for my motors. 

I found the grbl_ESP32_MPCNC controller card on tindie, and I think it can do what I want.  I don't want to use auto-squaring or have a ganged x axis, so can I just comment out the extra x step pins and the x and y axis squaring in cpu_map.h? 

```
#define USE_GANGED_AXES // allow two motors on an axis 
  
  #define X_STEP_PIN      GPIO_NUM_12
  // #define X_STEP_B_PIN    GPIO_NUM_22  // ganged motor
  // #define X_AXIS_SQUARING
  
  #define Y_STEP_PIN      GPIO_NUM_14
  #define Y_STEP_B_PIN    GPIO_NUM_21  // ganged motor
  // #define Y_AXIS_SQUARING
  
  #define Z_STEP_PIN      GPIO_NUM_27
  
  #define X_DIRECTION_PIN   GPIO_NUM_26
  #define Y_DIRECTION_PIN   GPIO_NUM_25  
  #define Z_DIRECTION_PIN   GPIO_NUM_33 
  
  // OK to comment out to use pin for other features
  #define STEPPERS_DISABLE_PIN GPIO_NUM_13
```

bdring, 2019-11-05:
It should work, so if it does not, I'll fix it. 

I won't be able to try it until late tomorrow

vertiginous, 2019-11-21:
In my initial post I mentioned I was going to use TMC2130's, but I was confused.  I actually ordered TMC5160 v1.2 drivers.   I was under the impression they were supported, but I got firmware support confused with support on this specific controller board. 

I have received my mpcnc controller from tindie, and I have it working with a few stepper motors and a4988 drivers.  When i plug in the TMC5160 drivers, the motors don't move.  I was under the impression these could be run in step\dir mode without any changes, but these boards are in SPI mode, and I've found no information on how to change them.  

My next thought was that I could add a few lines to my cpu_map and use SPI mode, and then after further reading I realized this board isn't designed for SPI mode.

I also considered using jumpers between the drivers and using the unused (in my case) X_STEP_B pin for the X_CS_PIN.  Now after another look at the schematic, I see that 3.3v is connected to those pins, so I'm thinking that won't work either.  I could use a little guidance if this combination of controller board and drivers is possible.

Thanks!

bdring, 2019-11-21:
The MPCNC controller does not support SPI mode. 

The ESP32 is very flexible, but limited in I/O pin count. It is hard to support too many options on a PCB, especially with 5 stepper drivers.

An MPCNC controller with SPI could be designed. I would suggest using SPI daisy chained drivers to reduce the number of pins required.

vertiginous, 2019-11-21:
OK, these drivers came with daisy chain jumpers, and I was hoping since I'm only using 4 drivers I could get creative with robbing pins from elsewhere.  I think the answer I'm looking for is somewhere between the mpcnc board and the 4 axis SPI board.  Thanks for your time!

bdring, 2019-11-21:
You could use those inverted pins to make it work. It would require soldering, but not cutting of traces.

You would need to repurpose a pin for the CS, like spindle enable. You would have to figure it our on your own, because I don't want to create a support problem for myself.

Bottom line: Grbl_ESP32 can do just about anything you need, but when you commit to copper, you need to make some permanent choices

vertiginous, 2019-11-22:
I don't think my soldering skills are up to the task.  I'm going to go with TMC2209's. They're working fine with no modifications.  Thanks again.  I learned a great deal from this exchange.

closed 2019-11-22 by vertiginous