bdring/Grbl_ESP32_MPCNC_Controller #8: Joint use of a relay and a PWM laser

maximius, 2020-04-07:
Repeated the grbl_esp 32_cnc_controller v1.2.1 Board. I Use a CNC engraver with the ability to connect a milling cutter and a laser. When using the relay to switch the router in the firmware we use pin 2 and the string
#define USE_SPINDLE_RELAY
to connect the laser to the PWM we use pin 17 and a string
//#define USE_SPINDLE_RELAY

Is it possible to configure the firmware so that the PWM is sent to pin 17, and the relay connected to pin 2 switched the relay simultaneously, without updating the esp32 firmware?
changing the parameter $32 (laser mode) changes only the strategy of switching on the relay, but not its pin.
This is important for combined processing on CNC machines. Thank you for any help.

maximius, 2020-04-07:
Good afternoon. Is it planned to implement the function of disabling the relay output when turning on laser mode $32=1

bdring, 2020-04-07:
You could use the spindle enable pin for the relay.

`#define SPINDLE_ENABLE_PIN      GPIO_NUM_2`

maximius, 2020-04-07:
I tried this option. When it is in the mode $32=0, the spindle works as it should, but

when entering the parameter $32=1 and flashing the code
//#define USE_SPINDLE_RELAY

#ifdef USE_SPINDLE_RELAY
#ifdef V1P1
#define SPINDLE_PWM_PIN GPIO_NUM_17
#else // V1p2
#define SPINDLE_PWM_PIN GPIO_NUM_2
#endif
#else
#define SPINDLE_PWM_PIN GPIO_NUM_16
#define SPINDLE_ENABLE_PIN GPIO_NUM_2
#endif

the relay is often switched on/off, but there is no PWM signal at output 16. If

//#define USE_SPINDLE_RELAY

#ifdef USE_SPINDLE_RELAY
#ifdef V1P1
#define SPINDLE_PWM_PIN GPIO_NUM_17
#else // V1p2
#define SPINDLE_PWM_PIN GPIO_NUM_2
#endif
#else
#define SPINDLE_PWM_PIN GPIO_NUM_16
#define SPINDLE_ENABLE_PIN GPIO_NUM_32
#endif

the PWM signal on the pwm output (pin 16) works as it should.

bdring, 2020-04-07:
I might be able to write a custom machine definition file. 

It would help to know exactly what you are trying to do and why.

Cows2Computers, 2020-07-08:
On the lowrider machine, allot of people have both a spindle and laser mounted at the same time. It would be nice to have the relay for turning on the spindle and PWM to run the laser. You would just then not supply source power to whichever tool you are not using.

Spacejamairways, 2023-11-07:
I made this well i tried 3 methods with videos. Linked pt1 https://youtube.com/shorts/0fDZFcwfVqU?si=npPupSD3e--cxUIT

Pt2 and coming. I burned my little red stepper driver and  I have tried to use one blue relay and then I tried 1 blue relay controlling a regulator big relay and then that didn't work so I tried high and low trigger relay which really didn't work . Going thru drivers alot hahah