bdring/Grbl_ESP32_MPCNC_Controller #9: Fan pin - faulty ground pin

indimini, 2020-05-01:
Hi, 

I just received my board and have grbl software installed on ESP32.  I have power to each of the stepper controllers,  but probing the fan pins w my multi-meter I see 0 V. if I move the gnd probe to the input power Gnd, and keep the + probe on the fan +5V pin, I see 4.95 V.

Is there a config setting I need to change in a .h file to get the fan post to work, or is this a fault in the PCB?

bdring, 2020-05-01:
The connector should be permanently connected to %V and Gnd. Can you check the solder connection on the back of the board?

[image]

indimini, 2020-05-01:
Visual inspection doesn't show anything obvious. If I check continuity from the solder joint on the board it still indicates a broken connection.

indimini, 2020-05-03:
I had no luck with getting gnd on that pin to work by resoldering, so I just soldered a jumper cable from the pin to the input gnd.  Looking at the picture above, it doesn't look like the fan gnd is connected to the input gnd - is that just on a different layer?

bdring, 2020-05-03:
There are only 2 layer. The bottom reddish layer is mostly a ground plane.  I do limited testing on each board, but every one gets a power supply test. I put power in the input terminal block and measure the output with a meter on those pins.

indimini, 2020-05-03:
The hard-wired jumper seems like it will work.  There was definitely no voltage across the pins prior to putting in the jumper.  Not a big deal - just thought I'd confirm it wasn't just a software config item before I broke out the soldering iron. :)  The rest of my tests have gone great.  I really appreciate you putting out this board as I was never a fan of Marlin for the CNC and this was a huge move forward.

bdring, 2020-05-03:
Sometimes those little pins have trouble soldering to a heavy ground plane.

closed 2020-05-23 by bdring