bdring/Grbl_ESP32_MPCNC_Controller #12: Questions/Issues re: homing

indimini, 2020-07-13:
I am trying to get the homing/squaring functions sorted out but am having some issues.  (I am only working with one axis at a time,  in this case, the X-axis.)  I have the steppers and limit switches wired up - movement to home is in the +X axis.  When I click on the Home X, the steppers run, the limit switches both trigger, but only one side will back off and reset the switch.  The other side will attempt to move in the +X direction and not back off from the closed limit switch. (The switches are wired as NC, so when closed, a continuity check shows no signal.)  Any insights on what I'm doing wrong or where my configuration needs adjustment would be appreciated.

bdring, 2020-07-13:
Start by making sure the switches are reading properly. Use this wiki page.

https://github.com/bdring/Grbl_Esp32/wiki/Help-with-Switch-Problems

indimini, 2020-07-16:
I've checked my switches and they work as expected.  Ensuring that the #10 bit 4 is set, if I run '?' I confirm that the pins work as expected.  When I home an axis (X for example), the gantry moves toward the limit switches, both switches trigger, the gantry backs off a bit then moves forward again.  At that point, one stepper motor will back off but the other stepper tries to move forward into the limiter.  It seems as if the instructions to back off is inverted for one o the two stepper motors.

bdring, 2020-07-16:
It might be easiest if we work together on this via slack. Is there a good time tomorrow? My day is open except after 5pm US central time

indimini, 2020-07-16:
That would be great, thanks!  I am East coast, I my only hard obligation is from 11 to 12 pm.  Let me know how to connect on Slack.

To add a bit more detail, whichever switch is triggered first is the one which will continue to move forward after first backing off the switch. The error is Alarm 6 - Homing fail. The active homing cycle was reset.  I've tried adjusting homing debounce (both increasing and decreasing) as well as the homing pull of (again increasing and decreasing) with no success.

Here's a summary of what I see on the status window for Pn:
Pn:YZ
Pn:YZ|WCO:10.000,0.000,0.000
Pn:XYZ|Ov:100,100,100
Pn:YZ
At that point, one stepper keeps moving in the wrong direction, triggering the ALARM 6: Homing fail. The active homing cycle was reset

indimini, 2020-07-21:
Are you still available sometime to track down this issue?  I have now confirmed the same behavior on both X and Y axis homing attempts.

bdring, 2020-07-21:
Yes, I live in the US Central time zone. Suggest a few reasonable times that work for you and we can interact on Slack.

indimini, 2020-07-22:
If you have time this Thurs or Fri between 11 am and 4 pm CT, that would work.  I can also do something later on Thursday.

bdring, 2020-07-22:
Friday at 2:00pm US Central time will work for me. Just have the machine ready to go and we can trouble shoot together.

indimini, 2020-07-22:
Great - how can I find you on slack?

bdring, 2020-07-22:
[Slack Link](https://join.slack.com/t/buildlog/shared_invite/zt-7twbithi-bRGScOaCgJoBpTbkykfxoA)

I will close this task after I see you join.

closed 2020-07-22 by bdring