bdring/Grbl_ESP32_MPCNC_Controller #14: How is the spindle relay being activated?

hgruber, 2020-10-19:
I got this far with this whole project (homing, plotting, ..), but I couldn't figure out how to enable/disable the spindle relay in grblesp. USE_SPINDLE_RELAY is defined in mpcnc_v1p2.h, but whatever M-code I use, the relay does not respond. It must be something stupid and simple... 

thanks for any quick help

bdring, 2020-10-19:
Use the most current version of the main or devt (beta) branches of the firmware. Use mpcnc_v1p2.h as your machine definition file.

In that file uncomment **#define USE_SPINDLE_RELAY**.  

After loading the firmware send **$Spindle/Type=RELAY** from the serial port of WebUI console. This only needs to be done once.

You should be able to activate the spindle relay with M3 S1 when in the idle state. If that does not work, please paste your startup messages here to help me debug. We can also help you in real time via Slack.

hgruber, 2020-10-22:
Okay, thanks, perfect. 
I also had to update to FW: 1.3a (20201004)
At first I thought homing was broken, only to find that it is even better.
Thanks for all that work, I really appreciate it!

closed 2020-10-22 by hgruber