bdring/Grbl_ESP32_MPCNC_Controller #17: port of your schematic

linewallker, 2021-09-11:
Hello,
  Some time ago I was trying to find ESP 32 CNC shield to buy and I found your schematic on the GITHUB so I downloaded your schematic and made some changes in it for my personal purposes.

Can I ask you for checking my schematic if it will be compatibile with your firmware ?

https://github.com/linewallker/ESP-32-CNC-SHIELD-PORT

If not I understnad.

Thank you for any kind of answer.

linewallker, 2021-09-24:
Hello, I am still waiting for your response.

bdring, 2021-09-24:
Sorry, I was on vacation when this message arrived. Please ask this
question on our  public channels of our Discord server. Please post a PDF
of the schematic and describe the changes you made. Also post a PDF of the
layout.

https://discord.gg/ArD5rPruKt

linewallker, 2021-09-26:
Hello,
  I have added Esp32_shield.pdf into repository with schematics.

Basicly I have made only small changes in one of your designs.

I have removed SD card because I am using big laser engraver 120 cm x 135 cm for photo engarving in full size and it makes files bigger than 2 gb so it is better to stream from PC.

I have cloned X and Y sockets for drivers, but not for independt driving of each motor, only created another socket with same connection as old one.

I have added pinout for external rellay, I have conected to EN circuit to take out signal for turn On and Off the rellay when the mottors are running, using for Air Assist compressor for 220V, I have already tested this connection on Arduino CNC shield connected to EN pin on unused Z axis and it is working fine with ferite filter on wiring to the rellay from shield.

Also I have alternated the connections on PCB but strictly kept your pinouts.

I hope this description is Ok for you :)

linewallker, 2021-09-26:
If you send me an email, I will share to you link on Altium Circuitmaker so you can see the full view if you want ;)

linewallker, 2021-09-26:
Try this :
https://365.altium.com/files/C485A942-C92D-11EB-A2F6-0A0ABF5AFC1B?openedFrom=files&variant=%5BNo%20Variations%5D

linewallker, 2021-09-28:
I have Updated detailed PDF file with schematics few minutes ago, I hope this will help.

https://github.com/linewallker/ESP-32-CNC-SHIELD-PORT/blob/main/esp32_shield.pdf